bozimmerman/Zimodem #153: Feature request: add zlib deflate stream decompression as a telnet option for MCCP2

danj2k, 2024-12-16:
As Zimodem natively supports standard telnet protocol, one of the possible use cases for it is connecting to a [MUD](https://en.wikipedia.org/wiki/Multi-user_dungeon).

Back in the early 2000s when computing power was becoming more readily available but Internet bandwidth was still at a premium, a protocol called MCCP2 was introduced which provides MUD servers with the ability to compress their outputs.

While bandwidth is less of an issue nowadays, many MUDs still support and actively use MCCP2, with some even providing additional benefits for players who enable it.

MCCP2 is implemented as a telnet option (86) and allows for communication from the server to the client (but not in the other direction) to take place in a zlib-compressed stream (e.g. see https://tintin.mudhalla.net/protocols/mccp/ for details).

It would not need the whole of zlib to be included, just a sufficient proportion to be able to inflate a DEFLATE-compressed stream. There already exist a few libraries which provide this kind of implementation on a microcontroller platform, for example [tinf](https://github.com/jibsen/tinf) or [udeflate](https://github.com/jlublin/udeflate).

bozimmerman, 2025-04-21:
Heh -- one of the OG use cases for this firmware, and the reason it has its fake telnet support, is precisely for MUDs.  This might be a good idea for the 4.x line, so thanks.

bozimmerman, 2025-04-21:
PS:  "Call" CoffeeMud @ coffeemud.net:23
:)