bozimmerman/Zimodem #156: What is the latest stable version of Zimodem?

eightbitswide, 2025-01-28:
I'm curious if 3.7.1 is really the latest stable version of Zimodem.
Poking around this folder: https://www.zimmers.net/otherprojs/  it looks like there are versions up to 4.44?   AT&U6502 updates to 3.7.1

Thanks

Biker1bob, 2025-04-11:
I wanted to ask the same. 3.7.1 is the last release on this. But the readme looks like it has been updated to 4.0 and Yes I looked in that folder as well and its at 4.44.

Anyone loaded up 4.44? is it good?  4.0??   Bo are you able to answer this for us?

bozimmerman, 2025-04-21:
So, it depends on your hardware.  The original GuruModem is still using the 3.7.x branch, as is the original C64Net-WiFi modems.  4.x is really for the new versions of the GuruModem with the updated storage and ram.  I've tested it with older ESP32s, however, and 4.x works fine with those also.  The only physical example of a 4.x modem was the new Guru-C64Net.  I don't know if ElectronicsIsFun is making updated Gurumodems yet.

closed 2025-04-21 by bozimmerman