bozimmerman/Zimodem #157: Amiga issue

Csikoskavics, 2025-01-29:
I have soldiered together a wifi modem using an ESP-12 for my Amiga 500.

Do I need to change some paramaters to work?
Compile itbdifferently?

What I getbis garbled characters. (Baud rate is correct also 8N1 is set.)

Thank you!

dabonetn, 2025-02-02:
How did you connect it?

Csikoskavics, 2025-02-02:
> How did you connect it?

Hello dabonetn,

Thx for checking on this!

I have found out about the issue.

My level shifter did not work correctly.

I think I will go with a MAX3237 instead.

Redesigned and ordered the pcb already.

Fingers crossed! :)

amiga23, 2025-02-03:
Which level shifter did you use? I tried TXS0108E and it does also not work.

Csikoskavics, 2025-02-08:
> Which level shifter did you use? I tried TXS0108E and it does also not work.
> 
> With the 4 channel Level Shifter with BSS138 from Adafruit it is working fine, but it only has 4 channles.

I have made my own, also out of some bss138.
It works nice on c64/128 and vc,-20, but not onmy Amiga 500. Also is quite compicated to soldier them.

Csikoskavics, 2025-02-18:
> Which level shifter did you use? I tried TXS0108E and it does also not work.
> 
> With the 4 channel Level Shifter with BSS138 from Adafruit it is working fine, but it only has 4 channles.

By te way! 8 channels: [https://www.ebay.com/itm/304709745626?_skw=Logic+Level+Converter+Module+Bi-Directional+Shifter+5V+to+3.3V+TTL+8&itmmeta=01JMBZ2PXZP3SJ7M9BBB0FQ4EQ&hash=item46f21dbbda:g:5LoAAOSwrRlZmAh7&itmprp=enc%3AAQAKAAABAFkggFvd1GGDu0w3yXCmi1fnguryZ4AeMsUzjASO502%2B6hW1jEwa1dvMmdqwKJxmWhxxECKE6aSymBSxIu%2Fk8UnjtdcoV6SDhG0XLQAyYEYE39YDmUHqWQfjWFrLNsBglbrkbtn9YsrW%2B7wMAdJeEGpDiP0pEIvNwLwNSJQgArwn2kDb8u%2FUB7BMMQNCdrmfzxQ0g%2B6w84ssBrzsT9ALRZ4MKbwC2aybf7WL4x2C3aC9YRApN%2FWZfKRMw0dAn34qnt%2BTIfM9GaqP414MpMcLBQw9M%2Fvgps7jsWoRRSsKidHM5lJeSehD4LFb814vyXRFUAKYabRLcPgepr%2FF2cg3WLM%3D%7Ctkp%3ABk9SR6bviv-iZQ](url)

dabonetn, 2025-02-25:
Looking at this tread again, the Amiga uses 1488/1489 line drivers for it's serial port, meaning full rs232 voltages , so simple logic level converters will not work. 

The Commodore 8 bit line used TTL levels for serial i.e. 0 to +5v, which is why those level converters work on them.

Google AI crap is still that, crap. It says that the a500 uses +5 and 0 volts, but looking at the schematics you can see the 1488 and 1489 line drivers connected to the serial port.

The MAX3237 that original poster replied with is a great choice for real rs232 interfaces.

Csikoskavics, 2025-03-06:
> Looking at this thread again, the Amiga uses 1488/1489 line drivers for it's serial port, meaning full rs232 voltages , so simple logic level converters will not work.
> 
> The Commodore 8 bit line used TTL levels for serial i.e. 0 to +5v, which is why those level converters work on them.
> 
> Google AI crap is still that, crap. It says that the a500 uses +5 and 0 volts, but looking at the schematics you can see the 1488 and 1489 line drivers connected to the serial port.
> 
> The MAX3237 that original poster replied with is a great choice for real rs232 interfaces.

Hello,

The new version of my modem is now Essambled. 

I can confirm, that as it was expected, MAX3237 works nice!

Now my problem became not the hardware, but software for on an Amiga 500! :D 

Thank you very much!

Csikoskavics, 2025-03-10:
Ok I was not 100% precise.

Did not verify flow control.

That I can't make it work. (RTS/CTS)

Do you know if i need to compile the code differenty might be for ESP8266?

What else settings I need, like AT&D2  might be?

dabonetn, 2025-03-10:
Did you use the config menu and enable it there?
AT+CONFIG

Csikoskavics, 2025-03-11:
> Did you use the config menu and enable it there? AT+CONFIG

Hi dabonetn,

Yes I did. 
Also using atf0 and there is a third method, but still, modem is not pulling cts high (Pin5 on the Amiga db25)

It looks to me that 
# define RS232_INVERTED
Has no effect somehow

Csikoskavics, 2025-03-11:
> > Did you use the config menu and enable it there? AT+CONFIG
> 
> Hi dabonetn,
> 
> Yes I did. Also using atf0 and there is a third method, but still, modem is not pulling cts high (Pin5 on the Amiga db25)
> 
> It looks to me that
> 
> # define RS232_INVERTED
> Has no effect somehow

Ohhhhh!

Lokk at this:

# define RS232_INVERTED 0

#ifdef RS232_INVERTED

It should be:

#if RS232_INVERTED == 0

I will test now

dabonetn, 2025-03-11:
If you have more issues with inverted, my link232 wifi repo has zimodem under it that uses noninverted on the esp8266, it defaults to inverted for the c64.

Csikoskavics, 2025-03-11:
> If you have more issues with inverted, my link232 wifi repo has zimodem under it that uses noninverted on the esp8266, it defaults to inverted for the c64.

Thank you very much dabonetn!

It was the inverted define, that have tricked me.

It looks like that I am able to push my modem now up until 76800 baud without going out of sync. (I have a Amiga 500 with 2mb fastram and 1mb chipram)

But I will need to do more precise tests.

closed 2025-04-21 by bozimmerman