bozimmerman/Zimodem #167: Pin Collision

petrkr, 2025-09-19:
https://github.com/bozimmerman/Zimodem/blob/86f7105602de5e3f1967a3a7e3e332226f8e3a07/zimodem/zimodem.ino#L123-L130

I creating new Modem Emulator board and I decided after 3 years to switch from ESP32 to ESP32 -S3.

So i trying to be compatible with default pinout, so I do not need to do Arduino board.json for it and it can be used with default DEV module.

But I noticed you have collision with RTS and RXD signals.

I guess in #define will won last one, so RXD will be 20.. So I suggest put RTS to pin 18 ? Or you have some different idea?

Mostly if I thining about pinout, I will take default HW pins and trying use them as much possible and since ESP32 family has more than one UART, then I reserve UART0 for debug outputs and UART 1 (2 respectively on ESP32) using for this kind of UART things. But that would be for different suggestion, like using Serial1 or Serial2 instead of HWSerial for ESP32* chips.

petrkr, 2025-09-30:
So I created PCB with following Pinout for ESP32-S3. 

If noone use ESP32-S3 on some hard-wired board but only duponts, it can be easily changed for DEV board as follows.

Bad is that Modem emuluator board is NOT part of Arduino boards.json, so we can not use "ifdef" unsless we will have some "#define MODEM_EMULATOR_BOARD_V4" somewhere at top of zimodem.ino file. For my testing I just fork it and rewritten constants for S3_DEV board.

For now I have tested just SD card, RS-232 I must get some terminal, but since I used same pinout (meaning name of pins) as on version 3 with ESP32, it should be working.

I also added two extra NeoPixel LEDs, so we can have more RGB states.. But also I kept backward compatible three GPIO LEDs for AA/HS/Wifi now.

Check it out, and if you want, I can send you some board for testing.

# Modem Emulator pinout

## RS-232
IO 15 - RS-232 RxD
IO 16 - RS-232 TxD

IO 17 - RS-232 RTS
IO 18 - RS-232 CTS

IO 04 - RS-232 DTR
IO 05 - RS-232 DCD
IO 09 - RS-232 DSR
IO 06 - RS-232 RI

IO 07 - RS-232 INVALID

## AUX header
IO 01 - AUX - pin 3 - I2C_SDA
IO 02 - AUX - pin 4 - I2C_SCL
IO 39 - AUX - Pin 8
IO 40 - AUX - Pin 7
IO 41 - AUX - Pin 6
IO 42 - AUX - Pin 5

## AUX2 Header
IO 21 - AUX2 - Pin 5
IO 47 - AUX2 - Pin 4
IO 48 - AUX2 - Pin 3

## LEDs
IO 03 - LED Red - Inverted
IO 45 - Led Green
IO 46 - Led Yellow
IO 38 - WS Led

## SD Card
IO 08 - SD Card Detect
IO 11 - SD MOSI (CMD)
IO 12 - SD SCLK (CLK)
IO 13 - SD MISO (DAT0)
IO 14 - SD CS (DAT2)

## Rest of GPIOs
IO 00 - Boot button

IO 10 - N/A

IO 19 - USB N
IO 20 - USB P

IO 35 - PSRAM
IO 36 - PSRAM
IO 37 - PSRAM

petrkr, 2026-01-22:
> " ESP32-S3 datasheed UART1 configured by IOMUX TX/RX are 17,18... " So, it sounds like the answer is: keep RTS at 20, because RX is already reserved at 18.

You can not keep gpio 20, because it is occupied by USB. And this can not be changed. You will totaly disable option to use usb cdc emulation for programming and enforce to use uart-usb converter. Which is pointless as this MCU support is by hardware, but on defined pins

bozimmerman, 2026-01-22:
OK, I'll take your word for it.  Your changes were merged upstream and are now in.